sparkfun/SparkFun_External_EEPROM_Arduino_Library #1: Does this library support writing type "String" ?

ullix, 2020-01-24:
Using an ESP32 with an 24C32 EEPROM I tried to write Strings to the EEPROM with these lines:

    `String myValue4[] = "this is my test string";   
    myMem.put(5, myValue4);   
    String myRead4;  
    myMem.get(5, myRead4); 
    Serial.printf("I read string: %s \n", myRead4.c_str());
    Serial.println("String putting and getting done");

` 
While put and get is done correctly as the prints give correct output, upon exiting the subroutine with this code, I get a "Corrupt Heap" crash right away.

Am I doing something wrong, or is this lib not supporting the String type?

nseidle, 2020-01-24:
Thanks for trying out the library! The String class is such a messy thing. In general this library will not work with Strings. I don't think any library (internal EEPROM or other) can handle Strings since the macros can't determine at compile time how long the string is. 

We could add a method that allows for variable length storage. Something like:

> String myValue = "Hello there";
char myChar[17];
myValue.toCharArray(myChar, 17);
myMem.put(5, myChar, 17); //Store 17 bytes at location 5

Would that work for you? It's still brittle because you have to make sure you don't overrun your arrays.

ullix, 2020-01-25:
Looks like I misinterpreted the examples to mean that all variable types are supported, which, however, is not the case for 'String's.  My workaround is using structs of char arrays, similar to what you suggested. That does it for me, and nothing further seems necessary.

I can certainly underwrite that this library is 'simple to use', very nice, thank you.

Hope I can pay my beer-duties some time ;-)

nseidle, 2020-01-25:
May we meet someday so I can have that beer! Thank you!

closed 2020-01-25 by nseidle