sparkfun/SparkFun_External_EEPROM_Arduino_Library #3: Write error across page boundary with AT24C512

dksmall, 2020-12-03:
I have a Mega2560 connected to the AT24C512 using the I2C interface. Pullups (2.1K) are on SDL and SDA.

I'm using Put to store a 17-byte structure to EEPROM memory, starting at address 1.  The first 3 records are fine, but the 4th record which should go to byte 68, stops writing at byte 63.  This is with a page size of 64.  Changing the page size to 128 will work until it hits byte 127, then stops writing.  If I write a single byte to any of these locations, that works as expected.  The only change to settings was going from 64 to 128 for the page size.

Code snippets:
struct eRecordType {
  float   Vout1;
  float   Vout2;
  float   Vout3;
  float   Vout4;
  byte    ValidRec;
};

eRecordType eRecord;          //variables used
ExternalEEPROM ExtMem;

ExtMem.put(addr,eRecord);   //where addr = 52 for page=64 or 120 for page=128

Attached are memory dumps where the same data is written until the page boundary is hit.
[memory dumps.txt](https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library/files/5638711/memory.dumps.txt)

closed 2020-12-05 by dksmall

nseidle, 2020-12-07:
Just curious - did you find a solution to the problem?

dksmall, 2020-12-07:
> Just curious - did you find a solution to your problem?

I was under the impression that I could do puts across a page boundary. But
I found comments in your source code which indicated we can’t do that. So I
changed my structure from 17 bytes to 32 bytes to make it page friendly. So
far this solution is working.  In hind sight I should have left a comment
on GitHub before I closed it.

We met way back in 2016 during AVC. I ran the unmanned rover event with a
GPS based Traxx RX truck and my son entered his first combat robot. We had
a blast then, too bad the event is no longer. 😩

Kelly Small

nseidle, 2020-12-07:
Hello Kelly!

Yep, page writes are a thing. It can be programmed around with some fanciness inside the library but the RAM footprint explodes. Glad you got it working!