sparkfun/SparkFun_External_EEPROM_Arduino_Library #10: Support for Arduino NANO 33 BLE SENSE (and any arduino mbed device)

alzzxx, 2021-02-11:
### Subject of the issue
Right now I'm using v1.03 of the library together with an Arduino nano 33 BLE sense with the latest version of the mbed core for arduino (Arduino nRF528x Boards (Mbed OS) v1.3.1) and I'm getting a compilation error because `I2C_BUFFER_LENGTH_RX and _TX` are not defined for this architecture:

`In file included from sketch\general_Head.h:38:0,

                 from c:\Users\a.rondon\Desktop\blackSabbath_encoder\Arduino\trunk\BlackSabbath\BlackSabbath.ino:2:

C:\Users\a.rondon\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src/SparkFun_External_EEPROM.h:59:19: error: This platform doesn't have a wire buffer size defined. Please contribute to this library!

 #pragma GCC error "This platform doesn't have a wire buffer size defined. Please contribute to this library!"

                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~`

### Your workbench
* Arduino nano 33 BLE sense
* Board version V2.0, with MCU nRF52840 (32-bit ARM® Cortex™-M4 CPU)
* EEPROM is wired to the board using I2C
* Power comes from USB port
* I'm using VScode with arduino plugin

### Steps to reproduce
Well it's simple, the code does not compile with the arduino nano 33 BLE (already tested the EEPROM and the library with Arduino Uno and Arduino Mega and both work OK), the error that I see is:

` #pragma GCC error "This platform doesn't have a wire buffer size defined. Please contribute to this library!"`

Looking at the library I can see that `I2C_BUFFER_LENGTH_RX and _TX` are not defined for mbed architecture:

`#if defined(ARDUINO_ARCH_APOLLO3)

#define I2C_BUFFER_LENGTH_RX AP3_WIRE_RX_BUFFER_LEN //Artemis is 256
#define I2C_BUFFER_LENGTH_TX AP3_WIRE_TX_BUFFER_LEN

#elif defined(__AVR_ATmega328P__) || defined(__AVR_ATmega168__)

#define I2C_BUFFER_LENGTH_RX BUFFER_LENGTH //I2C_BUFFER_LENGTH is defined in Wire.H
#define I2C_BUFFER_LENGTH_TX BUFFER_LENGTH

#elif defined(__SAMD21G18A__)

#define I2C_BUFFER_LENGTH_RX SERIAL_BUFFER_SIZE //SAMD21 uses RingBuffer.h
#define I2C_BUFFER_LENGTH_TX SERIAL_BUFFER_SIZE

#elif (defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MKL26Z64__) || defined(__MK64FX512__) || defined(__MK66FX1M0__) || defined(__IMXRT1062__)) // 3.0/3.1-3.2/LC/3.5/3.6/4.0

#define I2C_BUFFER_LENGTH_RX BUFFER_LENGTH //Teensy
#define I2C_BUFFER_LENGTH_TX BUFFER_LENGTH

#elif defined(ESP32)

#define I2C_BUFFER_LENGTH_RX I2C_BUFFER_LENGTH
#define I2C_BUFFER_LENGTH_TX I2C_BUFFER_LENGTH

#else

#pragma GCC error "This platform doesn't have a wire buffer size defined. Please contribute to this library!"

#endif`

I'd like to add something like:

`#elif defined(ARDUINO_ARCH_MBED)
#define I2C_BUFFER_LENGTH_RX SERIAL_BUFFER_SIZE
#define I2C_BUFFER_LENGTH_TX SERIAL_BUFFER_SIZE`

or:

`#elif defined(ARDUINO_ARCH_MBED)
#define I2C_BUFFER_LENGTH_RX BUFFER_LENGTH
#define I2C_BUFFER_LENGTH_TX BUFFER_LENGTH`

but honestly I have no idea how mbed Wire works, so I'm kindly asking for your help! :)

### Expected behavior
Well I'd expected that code will move on and not enter into the `#else #pragma` part

### Actual behavior
Goes into `#else` part and does not compile because `I2C_BUFFER_LENGTH_RX and _TX` are not defined

Any help will be greatly appreciated!

alzzxx, 2021-02-11:
OK, so after reading for a couple of hours I made an educated decision: the arduino nano 33 BLE is based on an arm MCU, so it uses ring buffer for Wire communication, similar to SAMD21G18A (also arm), so i decided to try this approach and add this:

```
#elif defined(__SAMD21G18A__) || defined(ARDUINO_ARCH_MBED)
#define I2C_BUFFER_LENGTH_RX SERIAL_BUFFER_SIZE //SAMD21 uses RingBuffer.h
#define I2C_BUFFER_LENGTH_TX SERIAL_BUFFER_SIZE
```

to line 42 of SparkFun_External_EEPROM.h

and now my code compiles, and I'm able to write and read from EEPROM, obviusly I'll continue to test it, but for now it's working!

nseidle, 2021-02-12:
"*after reading for a couple of hours I made an educated guess*" should be the title of all freshman engineering textbooks. Nice work! This is literally what my job entails every day :)

@alzzxx - Assigning the buffer size to the serial buffer size is ok but a bit dangerous. If they don't match you'll get all sorts of bad behavior.

@makin-stuff - Do you know the I2C buffer size of the nRF in Arduino and how to test for the nRF mbed platform?

For now, I've changed the hard pragma error to a warning. Please test v1.0.5.

closed 2021-02-12 by nseidle