sparkfun/SparkFun_External_EEPROM_Arduino_Library #16: Minor changes to memorySettings

wolfbert, 2022-10-31:
### Subject of the issue
1. Structure member `uint16_t i2cBufferSize` in SparkFun_External_EEPROM.h doesn't get initialized, which causes a compiler warning. Also, the member isn't used anywhere in the code, since the buffer size is #defined.
2. Structure member `uint32_t memorySize_bytes` is initialized to 512000 / 8. Assuming this is for for a 512Kbit EEPROM, shouldn't it be 512 * 1024 / 8 (i.e. 65536 bytes, or 64KB)?

### Your workbench
Arduino 2.0.1, ESP8266

### Steps to reproduce
1. Compile
2. `getMemorySize()` without setting it first

### Expected behavior
1. No compiler warning
2. Correct default memory size for Qwiic 512Kbit EEPROM (see [datasheet](https://www.onsemi.com/pdf/datasheet/cat24c512-d.pdf) and [product page](https://www.sparkfun.com/products/18355)), i.e. `getMemorySize()` returns 65536

[image]

### Actual behavior
1. SparkFun_External_EEPROM_Arduino_Library\src/SparkFun_External_EEPROM.h:149:3: warning: missing initializer for member 'struct_memorySettings::i2cBufferSize' [-Wmissing-field-initializers]
2. `getMemorySize()` returns 64000

### Proposal
1. Remove `uint16_t i2cBufferSize`: impacts existing code only if someone used the struct (which seems unlikely, as it's not used in the class interface). 
4. Change initialization as described: might affect existing code, probably not in a big way.

nseidle, 2022-10-31:
Sounds great! Feel free to modify and submit a PR as you see fit.

nseidle, 2022-11-04:
Thanks @wolfbert! In v1.0.9.

closed 2022-11-04 by nseidle