sparkfun/SparkFun_External_EEPROM_Arduino_Library #22: error in  byte size

fotosettore, 2023-06-23:
### Subject of the issue
I test my eprom 24C02 using Example1_BasicReadWrite.ino

### Your workbench
ESP32

### Expected behavior
the right size

### Actual behavior
I2C EEPROM example
Memory detected!
Mem size in bytes: 65536
I read: nan

nseidle, 2023-06-23:
Hi! The [24C02](https://www.digikey.com/en/products/detail/microchip-technology/24C02C-I-P/318616) is 2k bit or 256 bytes.

Does the Example 1 work correctly overall? Which line fails?

`myMem.length()` will return the internal variable [settings.memorySize_bytes](https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library/blob/master/src/SparkFun_External_EEPROM.cpp#L55) which is set to [64kB by default](https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library/blob/master/src/SparkFun_External_EEPROM.h#L159). You'll need to use the [setMemorySize()](https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library/blob/master/examples/Example2_Settings/Example2_Settings.ino#L48) function to set the size of the EEPROM you're using (256 in this case).

Hmm. I bet there's a way to auto-detect memory size. That will have to be a feature in the future!

fotosettore, 2023-06-23:
hi ... EXACT ! there must be a way to detect memory size automatically.
If we use setMemorySize() we lose the availability to found the exact type of eprom (example : when info written on it deleted or unreadable)

nseidle, 2023-06-24:
Please get the latest library version and see example 7. detectMemorySizeBytes() was fun to write! 

Note: detectMemorySizeBytes will write to the EEPROM and may cause data on the EEPROM to be overwritten! Please only use detectMemorySizeBytes() if there is nothing you need to read from the device.

I have tested it on [one board](https://www.sparkfun.com/products/18355) which means I have not really tested it at all. I will try to locate some additional EEPROMs for testing.

nseidle, 2023-07-04:
Ok, I've re-written a big portion of the library. Auto-detection is now working. Please see v2 of the library. I've successfully tested it on a 24LC01 and 24LC04. Please let me know if you continue to have issues with your 24C02.

closed 2023-07-04 by nseidle

fotosettore, 2023-07-04:
it's ok ... i'll try this day and i will give info about it