sparkfun/SparkFun_External_EEPROM_Arduino_Library #23: Latest version won't compile

amowry, 2023-07-05:
I'm seeing these errors with the latest version of the library. Do I need to change my previous sketches to make them compatible?Thanks!

`c:\Users\Andrew\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.cpp: In member function 'uint8_t ExternalEEPROM::detectWriteTimeMs(uint8_t)':
c:\Users\Andrew\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.cpp:202:29: error: invalid conversion from 'byte' {aka 'unsigned char'} to 'const uint8_t*' {aka 'const unsigned char*'} [-fpermissive]
  202 |         write(testLocation, magicValue, 1); // Uses polling
      |                             ^~~~~~~~~~
      |                             |
      |                             byte {aka unsigned char}
In file included from c:\Users\Andrew\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.cpp:25:
c:\Users\Andrew\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.h:122:55: note:   initializing argument 2 of 'int ExternalEEPROM::write(uint32_t, const uint8_t*, uint16_t)'
  122 |     int write(uint32_t eepromLocation, const uint8_t *dataToWrite, uint16_t blockSize);
      |                                        ~~~~~~~~~~~~~~~^~~~~~~~~~~
c:\Users\Andrew\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.cpp: In member function 'uint16_t ExternalEEPROM::detectPageSizeBytes()':
c:\Users\Andrew\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.cpp:363:21: warning: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '256' to '0' [-Woverflow]
  363 | #define maxPageSize 256 // Used in very large 2M bit EEPROMs
      |                     ^~~
c:\Users\Andrew\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.cpp:440:34: note: in expansion of macro 'maxPageSize'
  440 |         else if (pageSizeBytes = maxPageSize)
      |                                  ^~~~~~~~~~~`

nseidle, 2023-07-05:
Eeek! Sorry about that. Some platforms have more restrictive compiler settings.

Should be fixed in v2.0.1. I just tested some changes with ESP32, Uno, BLE Nano, Artemis, and a few others. Let me know if you still have issues and what board you're compiling for.

closed 2023-07-05 by nseidle

amowry, 2023-07-05:
That works-- many thanks!!