sparkfun/SparkFun_External_EEPROM_Arduino_Library #24: New version overwrites EEPROM by default?

amowry, 2023-07-05:
If I understand correctly, the new version overwrites the EEPROM by default in order to read the specs, is that correct? It seems a bit dangerous because updating to this version of the library will overwrite the EEPROM on a device with a previous version of the library, unless the settings were set manually before calling EEPROM.begin(). In my code I have been changing the settings after calling .begin(). Is that incorrect usage of the library?

nseidle, 2023-07-05:
No. The detection routines are written and tested to leave the original data unaltered. I had some original notes that had warnings but those should all be removed. Please see example 8 for a test of writing data, then running detection routines, then verifying the original data is intact.

The correct usage: generally, if you know the settings of your eeprom, set them before begin(). 

Please keep the questions coming and let me know if documentation needs updating.

gerrygralton, 2023-07-28:
Hi, 
I'm just going to jump in on this. I'm using a M24C64 EEPROM chip and the library seems to work pretty well but it was overwriting data. Originally the setup was:
```
EEPROM.setMemorySizeBytes(memory_size);
EEPROM.setPageSizeBytes(page_size);

if (EEPROM.begin(EEPROM_add, WireI2C1) == false){
    error(2);
}
```
and data was being consistently overwritten. I added one line:
```
EEPROM.setAddressBytes(2);
EEPROM.setMemorySizeBytes(memory_size);
EEPROM.setPageSizeBytes(page_size);

if (EEPROM.begin(EEPROM_add, WireI2C1) == false){
    error(2);
}
```
and it now works perfectly. I agree with @amowry that this is dangerous behaviour. At the very least, it should be stated somewhere in the README how many things need to be set to stop this autodetection stuff.

nseidle, 2023-07-29:
Ya, I now agree. I'll remove the detection bits from the begin function and move to v3.

nseidle, 2023-07-29:
Ok, v3 is up with all the autodetection removed from begin(). There's probably other confusing bits in the examples but I wanted to get the release out. Please have a look and let me know if anything else needs clean up.

amowry, 2023-07-31:
V3 doesn't seem to be working correctly for me. It seems that it may be reading and writing to/from different locations in EEPROM. When I have a few more minutes I'll try to figure out exactly what's happening.

amowry, 2023-09-20:
I still haven't been able to get v. 3.0.0 to work correctly. Has anyone else had this issue? I'm having to revert to 2.0.1.

I've tried calling begin() both before and after changing the settings, like in the two examples below, but the library seems to be reading and writing from different places in the EEPROM.

`    //I2C
    Wire.begin();                          //Join i2c bus for EEPROM.
    Wire.setClock(400000);                 //high speed
    EEPROM.setMemorySize(128 * 1024 / 8);  //In bytes. 128kbit = 16kbyte
    EEPROM.setPageSize(64);                //In bytes
    EEPROM.enablePollForWriteComplete();   //Supports I2C polling of write completion. This shortens the amount of time waiting between writes but hammers the I2C bus by polling every 100 uS. disablePollForWriteComplete() will add 5 mS between each write.
    EEPROM.setPageWriteTime(5);            //5 ms max write time
    EEPROM.begin();                        //Start M24128 EEPROM communication using the default address of 0b1010000`

`    //I2C
    Wire.begin();                          //Join i2c bus for EEPROM.
    Wire.setClock(400000);                 //high speed
    EEPROM.begin();                        //Start M24128 EEPROM communication using the default address of 0b1010000
    EEPROM.setMemorySize(128 * 1024 / 8);  //In bytes. 128kbit = 16kbyte
    EEPROM.setPageSize(64);                //In bytes
    EEPROM.enablePollForWriteComplete();   //Supports I2C polling of write completion. This shortens the amount of time waiting between writes but hammers the I2C bus by polling every 100 uS. disablePollForWriteComplete() will add 5 mS between each write.
    EEPROM.setPageWriteTime(5);            //5 ms max write time`

nseidle, 2023-09-21:
I've rolled the release (v3.0.1) back to v1.0.14 until I get this sorted.

nseidle, 2023-09-21:
If you don't mind, I'd like to track this issue. I'll edit the title to point at the v3 branch.

amowry, 2023-09-21:
Of course, thanks!

nseidle, 2023-10-27:
Ok, to close this out: 

* v3.2.x is pretty stable and well tested against 24xx00 to 24xx1025 model EEPROMS.
* Autodetection is turned off by default. But is still available for those that need it in Example 4. I'm not 100% sure it maintains previously recorded data when used, but I'm pretty sure.
* Either way, see Example 1 and use setMemoryType() to tell the library the EEPROM you've got.

closed 2023-10-27 by nseidle