sparkfun/SparkFun_External_EEPROM_Arduino_Library #25: The library doesn't read from the correct location in EEPROM

amowry, 2023-09-20:
### Subject of the issue
With v, 3.0.0, the library seems to read from the wrong location in EEPROM.

### Your workbench
I'm using a custom NRF52840 board with the Adafruit ItsyBitsy NRF52840 bootloader and board package. The EEPROM is an ST M24128-BFMH6TG.

The previous of the library (2.0.1) works fine with this setup.

### Steps to reproduce
Using the attached code and library v. 2.01, the value "10" is read back from address 1 in EEPROM, as would be expected.
However, with v. 3.0.0, a value of "255" is returned, indicating that the library is reading from some other address than 1.

[Sparkfun_external_EEPROM_library_test.txt](https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library/files/12678517/Sparkfun_external_EEPROM_library_test.txt)

toogooda, 2023-09-21:
Yes please roll back V3 it is very broken waist so much time then tried previous library and all is working just get 255 for reads

amowry, 2023-09-21:
v1.0.14 seems to work well for me. Thanks for your help!

nseidle, 2023-09-21:
Glad to hear it's working for you. Sorry for introducing a bad version!

amowry, 2023-09-21:
No problem at all-- thanks for your support!

nseidle, 2023-10-09:
Sorry, I don't know how to down-force PlatformIO.

amowry, 2023-10-09:
I've actually noticed the same thing on Arduino IDE-- it still says that 3.0.0 is the most recent version. Should that be showing 3.1 as an option? I've manually selected 1.0.14 and haven't been letting it update.

pauljeffress, 2023-10-12:
@nseidle @amowry yeah I think perhaps the only solution is to create v3.1 with the known working code from 1.0.14. Because as far as I know both Arduino IDE and PlatformIO just look at GitHub, so and if a new person installs or updates the library today they will get the broken v3.0.

nseidle, 2023-10-12:
A fine hack to the problem. New v3.1.0 released. Please let me know if you see it listed correctly.

amowry, 2023-10-12:
The 3.1.0 update appears in Arduino IDE now and seems to work fine. Thanks!

amowry, 2023-10-26:
Both 3.1.1 and 3.1.2 seemed to break this again-- it's reading all 255 values from the EEPROM.

nseidle, 2023-10-27:
Sorry for breaking things again, again.

I've pushed v3.2.0. Please have a test.

I had a 'ah ha' moment, let's see if it's good...

    myMem.setMemoryType(128); // Valid types: 0, 1, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048

I've added a setMemoryType() to make initialization (no auto-detect) more straight forward. 

I've tested this against the 24xx128 an everything works as expected in Ex7. Please let me know how to replicate your setup.

[image]

pauljeffress, 2023-10-27:
@nseidle 3.2.0 is failing for me, I rolled back to 3.1.0 and its was all good.
I have a AT24C32 (32kx8bit i.e. 4096 Bytes), with addressing set to 0x57, not the default 0x50.
I'm using a SAMD51 MCU.
I have a slight variation of your test code that I am using for the test with diff library versions. 
Note, the memory size that's detected is always wrong, I think it always has been.
**Results with 3.1.0**
```
17:50:20.424 -> Qwiic EEPROM example
17:50:20.424 -> Memory detected!
17:50:20.424 -> Mem size in bytes: 65536
17:50:20.424 -> I read: 200
17:50:20.424 -> I read: -366
17:50:20.424 -> I read: -7.35
```
**Results with 3.2.0**
```
17:38:17.950 -> Qwiic EEPROM example
17:38:17.950 -> Memory detected!
17:38:17.950 -> Mem size in bytes: 65536
17:38:17.950 -> I read: 0
17:38:17.950 -> I read: 0
17:38:17.950 -> I read: 0.00
```
**Results with 1.0.14**
```
17:52:44.395 -> Qwiic EEPROM example
17:52:44.395 -> Memory detected!
17:52:44.395 -> Mem size in bytes: 65536
17:52:44.395 -> I read: 200
17:52:44.395 -> I read: -366
17:52:44.395 -> I read: -7.35
```

pauljeffress, 2023-10-27:
My code for the above testing is;

`
/*
 * Based massively on "Example1_basicreadwrite.ino", the example that comes with the 
 * "Sparkfun External EEPROM" Library.
 * 
 * This worked when I had my ZS-042 board with RTC+EEPROM connected to GCM4.
 * 
 * My EEPROM is at 0x57, not the default of 0x50
 * 
 * 
 */

/*
  Read and write settings and calibration data to an external I2C EEPROM
  By: Nathan Seidle
  SparkFun Electronics
  Date: December 11th, 2019
  License: This code is public domain but you buy me a beer if you use this 
  and we meet someday (Beerware license).
  Feel like supporting our work? Buy a board from SparkFun!
  https://www.sparkfun.com/products/14764

  This example demonstrates how to read and write various variables to memory.

  The I2C EEPROM should have all its ADR pins set to GND (0). This is default
  on the Qwiic board.

  Hardware Connections:
  Plug the SparkFun Qwiic EEPROM to an Uno, Artemis, or other Qwiic equipped board
  Load this sketch
  Open output window at 115200bps
*/

#include <Wire.h>

#include "SparkFun_External_EEPROM.h" // Click here to get the library: http://librarymanager/All#SparkFun_External_EEPROM
ExternalEEPROM myMem;

void setup()
{
  Serial.begin(115200);
  delay(2000);
  Serial.println("Qwiic EEPROM example");

  Wire.begin();

  if (myMem.begin(0x57, Wire) == false)
  {
    Serial.println("No memory detected. Freezing.");
    while (1)
      ;
  }
  Serial.println("Memory detected!");

  Serial.print("Mem size in bytes: ");
  Serial.println(myMem.length());

  //Yes you can read and write bytes, but you shouldn't!
  byte myValue1 = 200;
  myMem.write(0, myValue1); //(location, data)

  byte myRead1 = myMem.read(0);
  Serial.print("I read: ");
  Serial.println(myRead1);

  //You should use gets and puts. This will automatically and correctly arrange
  //the bytes for larger variable types.
  int myValue2 = -366;
  myMem.put(10, myValue2); //(location, data)
  int myRead2;
  myMem.get(10, myRead2); //location to read, thing to put data into
  Serial.print("I read: ");
  Serial.println(myRead2);

  float myValue3 = -7.35;
  myMem.put(20, myValue3); //(location, data)
  float myRead3;
  myMem.get(20, myRead3); //location to read, thing to put data into
  Serial.print("I read: ");
  Serial.println(myRead3);
}

void loop()
{
}

`

nseidle, 2023-10-27:
Your code does not have any memory initializers so [the defaults are applied](https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library/blob/main/src/SparkFun_External_EEPROM.h#L184):

        .memorySize_bytes = (uint32_t)512 * 1024 / 8, // 65,536 bytes = 512-Kbit EEPROM
        .pageSize_bytes = 64,
        .addressSize_bytes = 0,

The default of addressSize_bytes of 0 is likely the culprit. Previous versions (v1.x) calculated the addressSize_bytes on the fly which meant is was not able to handle all EEPROM types. I'll change this to 2 bytes size I'm focused on the larger EEPROMs (such as the Qwiic board) that use a >2048 byte (model >24xx16) EEPROM.

Add the following line to your code above myMem.begin();

    myMem.setMemoryType(32);

This will set the memory size, page size, and address size correctly. This was a breaking change from v1 to v3.

amowry, 2023-10-27:
Okay, I just tested it. For me, 3.2.0 fails if I use setMemorySize but succeeds if I use setMemoryType.

3.1.2 fails, and 3.1.0 succeeds. I used the BasicReadWrite example, with the only change being switching between setMemorySize and setMemoryType:

`    //myMem.setMemoryType(128);  // Valid types: 0, 1, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048
    myMem.setMemorySize(128 * 1024 / 8);  //In bytes. 128kbit = 16kbyte` 

I'm using an Adafruit NRF52840 Express with M24128-BFMH6TG EEPROM. 

Here are my results:

3.1.2:

Qwiic EEPROM example
Memory detected!
Mem size in bytes: 16384
I read (should be 200): 255
I read (should be -366): -1
I read (should be -7.35): -nan

3.2.0 with setMemorySize:

Qwiic EEPROM example
Memory detected!
Mem size in bytes: 16384
I read (should be 200): 255
I read (should be -366): -1
I read (should be -7.35): -nan

3.2.0 with setMemoryType:

Qwiic EEPROM example
Memory detected!
Mem size in bytes: 16384
I read (should be 200): 200
I read (should be -366): -366
I read (should be -7.35): -7.35
I read: Hi, I am just a simple test string
Next available EEPROM location: 65

3.1.0:

Qwiic EEPROM example
Memory detected!
Mem size in bytes: 16384
I read (should be 200): 200
I read (should be -366): -366
I read (should be -7.35): -7.35
I read: Hi, I am just a simple test string
Next available EEPROM location: 65

amowry, 2023-10-27:
If setMemorySize is deprecated I'm happy to use setMemoryType and move on :). Thanks for your help!

nseidle, 2023-10-27:
> If setMemorySize is deprecated

Not so much deprecated, as not needed or recommended. I recommend using setMemoryType or use the combination of setMemorySizeBytes, setAddressBytes, and setPageSizeBytes. See example 5 (new in latest release).

> happy to use setMemoryType and move on :)

Yep, I agree. It's time for me to stop breaking this. Thanks for your patience!

I'm going to close this issue since the OP is working. If folks have other issues, please open a new one.

closed 2023-10-27 by nseidle