sparkfun/SparkFun_External_EEPROM_Arduino_Library #26: Example #1 does not work with AT24C02 eeprom

pkhardage, 2023-10-23:
### Subject of the issue
Example #1 does not run after downloading library from Github and compiling and uploading successfully. AT24C02 is only device connected to I2C bus. I2c scanner identifies AT24C02 with address 0x50. SparkFun tested SW with 1 mbit and 500 kbit memories and I suspect the AT24C02 at 2 kbits is not tested or not supported.

### Your workbench
Using my SW tests board with DFROBOT Beetle and Arduino IDE and works with other I2C libraries. Proven hardware system.
* What development board or microcontroller are you using?
* What version of hardware or breakout board are you using? 
* How is the breakout board wired to your microcontroller?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Use example #1, download library, compile-download, should see some serial monitor output but none.

### Expected behavior
Should at least print first Serial.print statement in SETUP.

### Actual behavior
Tell us what happens instead

nseidle, 2023-10-24:
24LC02s should be working just fine:

    // 24LC02 - 2048 bit / 256 bytes - 1 address byte, 8 byte page
    myMem.setAddressBytes(1);
    myMem.setPageSizeBytes(8);
    myMem.setMemorySizeBytes(256);

[image]

I test on all EEPROM sizes (ok, except LC256).

[image]

I'm going to close the issue because it's working for me. Please continue to post and I'll try to help you troubleshoot.

nseidle, 2023-10-24:
Hold on, I may have been on a bad branch. Let me re-test.

nseidle, 2023-10-24:
Alright, please give v3.1.1 a try and let me know what problems you see.

pkhardage, 2023-10-24:
Thanks nseidle,  I am new to the Github environment so I see you released a v3.1.1 eleven hours ago but I only see v3.1.0 as available for install. How do I get to v3.1.1 to try even if it is not ready for public release?

pkhardage, 2023-10-24:
Ok, will install and let you know result.  Thanks

pkhardage, 2023-10-24:
Lib manager still showing v3.1.0 as latest release.

pkhardage, 2023-10-24:
SparkFun_External_EEPROM examples no longer showing in Arduino IDE.

pkhardage, 2023-10-24:
> Dunno what's up with the Arduino lib manager. It's showing up here https://downloads.arduino.cc/libraries/library_index.json.
>
> For the interim, please install via zip method https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries#importing-a-zip-library.

Nathan,  I will be out for a while.  The v3.1.1 has not shown up and the SparkFun_External_EEPROM library examples have disappeared.
Only the SparkFun VL53L1X Laser Dist. Sensor is left in examples for SparkFun.
Don't know why but happened with changes on Github made this morning.
I will check later today when I get back home.

pkhardage, 2023-10-24:
3.1.2 up now? Do I have to select which memory chip I am compiling for and uncomment those lines under Wirebegin()??
Are there user directions on "how to use" that I am missing?

pkhardage, 2023-10-24:
I have downloaded v3.1.2 and the new example 1 file but still have no results. What do I do next?

nseidle, 2023-10-24:
Did you modify Ex1 to the following:

    // 24LC02 - 2048 bit / 256 bytes - 1 address byte, 8 byte page
    myMem.setAddressBytes(1);
    myMem.setPageSizeBytes(8);
    myMem.setMemorySizeBytes(256);

pkhardage, 2023-10-24:
Yes, but I think I have found the problem in the examples.  There is only a void setup() and no void loop() so there is just one pass through the example. You have to get the Serial Monitor going before you exercise the rest of the setup. If Serial Monitor is not up then you see no output. Adding a delay of 2000 msec after Serial.begin gets the monitor up in time for the rest of the  serial.print statements. i.e.
Serial.begin(115200);
  delay(2000);
  Serial.println("Qwiic EEPROM example");
I think all the examples need to address this issue.

nseidle, 2023-10-25:
Ah! Yep, that's common with ESP32 setups. Glad to hear it's working!

closed 2023-10-25 by nseidle

pkhardage, 2023-10-26:
Hi Nathan, thanks for the help and release of External_EPROM library to fix my issues. This External_EEPROM library is great that it mirrors the Internal EEPROM
library. I do see additional functionality in the External_EEPROM library i.e
nextEEPROMLocation which could be very useful. Is there any documentation on the additional functions. When I try to apply the nextEEPROMLocation function, it always comes back with address 0. Any help would be appreciated.
 
I will try to get you some user perspective feedback on all the examples soon if you are interested.
 
Thanks again,  KentH