sparkfun/SparkFun_External_EEPROM_Arduino_Library #28: Missing settings for 1Mb EEPROMS (131072 words, 256-byte page)

dzalf, 2023-11-08:
Hi guys

Settings for the [CAT24M01](https://www.onsemi.com/pdf/datasheet/cat24m01-d.pdf) EEPROM are not available.

The closest available configuration is: `myMem.setMemoryType(1025)` however the indicated memory size is 128000 while the CAT24M01 is 131072 words of 8 bits. (Or am I missing something? 😵‍💫

nseidle, 2023-11-08:
100 year data retention! Neat. I have not seen this one before.

See [example 5](https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library/blob/main/examples/Example5_CustomSettings/Example5_CustomSettings.ino). I suspect it will work with custom settings. Something like:

    //Explicitly set the address bytes, page size, and memory size for this EEPROM
    myMem.setMemorySizeBytes(131072);
    myMem.setAddressBytes(2);
    myMem.setPageSizeBytes(256);

Let me know if that doesn't work. I may have to order a unit to test.

dzalf, 2023-11-08:
> 100 year data retention! Neat. I have not seen this one before.
>
> See [example [...]

This makes a lot of sense. I had the feeling that there should be some programmatic options like that but I was so sleepy last night when I posted my _issue_ that I did not read the codebase thoroughly.

I will test it and report my results back to you.

_100 year data retention! Neat. I have not seen this one before._
 Yeah, good enough for leaving a digital time capsule that only people with EEPROM readers will be able to open in the year 2123 😮

nseidle, 2023-11-15:
I'm going to assume the custom settings worked for the [CAT24M01](https://www.onsemi.com/pdf/datasheet/cat24m01-d.pdf) so I'm closing for now. Please let us know if you still have problems.

closed 2023-11-15 by nseidle

dzalf, 2023-11-16:
Yes, @nseidle 

Sorry for not getting back to you earlier. I am still running some tests. I will come back in case I need some help.

My original and _very n00b-ish_ plan was to use the EEPROM as an alternative for more RAM but I am concerned about wearing out the chip after several read-writes

nseidle, 2023-11-16:
No worries. Depending on your application, EEPROM may or may not work as RAM. EEPROM is *way* slower to write to than RAM, but generally wear is usually not a terrible concern. If you're building something for life-safety applications, then yes, be very concerned. If you're building a DIY project, don't worry about it.

See our [Flash Thrasher](https://github.com/nseidle/Flash_Thrasher/blob/master/readme.md) repo. Note this was for flash inside the ATMega328, so a very different application, but in general, the datasheets for EEPROM (flash, etc) are very conservative and you'll see 10 or 100x what the datasheet quotes. This is why I'm so impressed they quoted 100 year retention - to me that means that every chip they produce will last 100 years but most will last much much longer.