sparkfun/SparkFun_External_EEPROM_Arduino_Library #29: Erros with SparkFun Qwiic EEPROM Breakout - 512Kbit

RemyDavies, 2023-11-08:
Hi all, can I please get your thoughts on this?

I'm running a SparkFun Qwiic EEPROM Breakout - 512Kbit (https://www.sparkfun.com/products/18355). It's been running for months without issue with my Arduino MKR Zero, until recently when I updated the library to v3.2.1

I've tried rolling back through previous versions but then I experience other issues.

When I run the "**Example8_InterfaceTest.ino**" - it throws many 'fails', followed by "Something went wrong. See output".

[image]

Similarly, the **"AutoDetectionTest.ino"** also returns an error in location 254.
[image]

nseidle, 2023-11-14:
Sorry for the hassle! A lot changed over the last few months but everything is pretty stable at the moment.

I've tested Ex 1 and Ex 8 using the SparkFun Qwiic EEPROM and everything works as expected.

[image]

[image]

[image]

What is the output when you run Ex1? If you don't see output on Ex1, uncomment the delay(250) at the top of setup().

> When I run the "Example8_InterfaceTest.ino" - it throws many 'fails', followed by "Something went wrong. See output".

Did you set the memory type? 

    myMem.setMemoryType(512); // Valid types: 0, 1, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048

Otherwise, it defaults to 24xx32, not 24xx512.

RemyDavies, 2023-11-14:
Thanks for the reply! Still no luck (see below). 

I'm not missing something obvious here am I? Not sure whats going on.

Result from Ex1:
[image]

Result from Ex8, now with setMemoryType(512):
[image]
[image]

Result from Ex8, now with setMemoryType(512) and setMemorySizeBytes(512):
[image]
[image]

nseidle, 2023-11-15:
You are setting the memory type correctly, then you are overwriting it with these lines.

    myMem.setMemorySizeBytes(256);
    myMem.setAddressBytes(1); //Set address bytes and page size after MemorySizeBytes()
    myMem.setPageSizeBytes(1);

 You need one (setMemoryType) or the other (setMemorySizeBytes/setAddressBytes/setPageSizeBytes) but not both.

closed 2023-11-15 by nseidle

RemyDavies, 2023-11-15:
Thank you! Ex8 passes now :)

Just to clarify - when using this EEPROM library in future, are there any other parameters (besides 'setMemoryType') that I need to initialise in setup?

I'd just like to be sure since setMemoryType wasn't required in previous versions, etc. :)

Also, any chance you may know why I'm seeing the following compile warnings?

E:\OneDrive\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.cpp: In member function 'uint32_t ExternalEEPROM::detectMemorySizeBytes()':
E:\OneDrive\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.cpp:593:13: warning: unused variable 'i2cAddress' [-Wunused-variable]
     uint8_t i2cAddress = settings.deviceAddress;
             ^~~~~~~~~~
E:\OneDrive\Documents\Arduino\libraries\SparkFun_External_EEPROM_Arduino_Library\src\SparkFun_External_EEPROM.cpp:626:9: warning: unused variable 'loopCount' [-Wunused-variable]
     int loopCount = 1;

nseidle, 2023-11-15:
Please give v3.2.2 a try and let me know if there's any other problems (in a new issue).

commit d6ed4c38 (2023-11-15): Remove unused variables

Fix #29