sparkfun/SparkFun_External_EEPROM_Arduino_Library #36: Return type of put and get.

SpintroniK, 2024-05-02:
### Subject of the issue
Just wondering why the return type of `put` is `T&`? (same for `get`).
I don't really see why these functions return the value we feed them with.
I wonder if it wouldn't make more sense to have `put` return the next writable location, kind of like `putString`.
That way, it would become possible to save structs one after another, by using the returned value as the next address.
Perhaps it would also make sense if `get` just returned void.
What do you guys think?

### Your workbench
* Not relevant.

### Steps to reproduce
Not relevant.

### Expected behavior
Not relevant.

### Actual behavior
Not relevant.

nseidle, 2024-07-11:
Sorry for the late reply. I suspect the return type is just a hold over from the Arduino internal eeprom library (what this library is based on).

I'm very open to changing the return type for put. I doubt many folks are using it so I'm ok with a breaking change.

SpintroniK, 2024-07-17:
Hi, no worries.
I think it would be a lot nicer if `put` returned the next writable address.
That'd be quite useful, in fact, as it would allow to chain calls to `put` quite easily.