sparkfun/SparkFun_External_EEPROM_Arduino_Library #37: Option to make bulk write only write changes

merlinz01, 2024-07-10:
### Subject of the issue
Currently when you call `put()` with a struct, the entire struct is written to the memory whether or not anything changed. Could we have a method that checks it byte-by-byte and writes only the changed bytes, in order to preserve the EEPROM lifetime?

Or does the EEPROM do this internally?

### Your workbench
* What development board or microcontroller are you using? N/A
* What version of hardware or breakout board are you using? 24C02
* How is the breakout board wired to your microcontroller? I2C
* How is everything being powered? N/A
* Are there any additional details that may help us help you? N/A

### Steps to reproduce
Call `eeprom.put(MyStruct)`.

### Expected behavior
There should be a method or parameter that does the same thing but checks for changes so that we don't put unnecessary wear on the EEPROM.

### Actual behavior
The entire struct is written whether or not it changed.

merlinz01, 2024-07-10:
I have opened a PR with an implementation.

nseidle, 2024-07-12:
This is wonderful! Thank you for your contribution.

closed 2024-07-12 by nseidle