sparkfun/SparkFun_External_EEPROM_Arduino_Library #42: The Qwiic EEPROM does not work with the RedBoard Qwiic

jim-rothrock, 2024-10-29:
I have tried two different Qwiic EEPROMs (item COM-18355) with two different RedBoard Qwiics (item DEV-15123) with two different Qwiic cables. In all cases, the Example1_BasicReadWrite sketch prints “Mem size in bytes: 0” and “I read (should be 200): 255”. When I use either EEPROM with an Adafruit QT Py M0 (SAMD21), the correct values (i.e. 65536 and 200) are printed. The Qwiic EEPROM Hookup Guide shows the Qwiic EEPROM being used with a RedBoard Qwiic, so that combination of hardware should be supported.

When I run an I2C scanner, I get “I2C device found at address 0x50”, which is correct. Only one EEPROM is connected to the RedBoard at a time.

I ran the example sketch on an Arduino Uno R2 with a SparkFun BOB-08745 level converter (10 k pull-ups) between the Arduino and the EEPROM. As with the RedBoard Qwiic, the memory size is detected as 0 and the write of 200 does not work (i.e. the Arduino acts as if the write didn’t happen). I then cut the pull-up jumpers on the EEPROM, thinking that the I2C pull-up might be too strong, but there was no change in behavior.

john.trepke at SparkFun duplicated the problem.

[SparkFun Forum post](https://community.sparkfun.com/t/qwiic-eeprom-does-not-work-with-redboard-qwiic/59682)

nseidle, 2024-10-30:
Hi Jim - Thanks for reporting and sorry for the hassle. I don't have my test gear with me at the moment but hope to have a look tonight.

nseidle, 2024-10-31:
Everything is working more or less. setMemoryType() is not casting typeNumber to a uint32_t so the size is getting set incorrectly. Here's the output with the fix in place.

    Qwiic EEPROM example
    Memory detected!
    Mem size in bytes: 65536
    I read (should be 200): 200
    I read (should be -366): -366
    I read (should be -7.35): -7.35
    I read: Hi, I am just a simple test string
    Next available EEPROM location: 65

I'll get a version released shortly.

Now why is the string not fully printing?

nseidle, 2024-10-31:
Please get v3.2.9 when it's made available (usually less than an hour). You should be all set but let us know if you hit any other hurdles.

closed 2024-10-31 by nseidle

jim-rothrock, 2024-11-06:
I have verified that version 3.2.9 fixes the problem. Thanks!

commit bb73a2d0 (2024-10-31): Cast typeNumber to uint32_t

Fix #42. Not all compilers were casting. AVR specifically was failing.