sparkfun/SparkFun_External_EEPROM_Arduino_Library #44: wrong data when using arrays in struct

herbert131, 2025-07-23:
I'm using a struct (~272 byte) to manage my data. withing the struct there are two tow-dimentional float array each 128 Byte.
when I read out the struct, some data (always [3][4] of one array is corrupt - showing "-inf". 
When I access the dedicated corrupt entry directly the value is read out correctly. ( reading single bytes in the suspected areas did als not show any wrong data ...
Are there limitations when using arrays ?  

```
typedef struct                
{
    uint8_t version;             // 1 byte
    uint8_t reserved[3];       // 3 Byte  padding

    bool   cal_A[4];      // 4 Byte
    bool   cal_B[4];      // 4 Byte
    float   korr_B[4][8];   // 128 Byte
    float   korr_A[4][8];   // 128 Byte

    uint16_t reserved2;        // 2 Byte
    uint16_t crc;              // 2 Byte        
} calstruct;
```

// TEST read out [3][0] to [3][7]           -  > value of [3][4]  corrosponts to test array[4] is correct
    ```
 float   mem_valuefloat;
 float   test_array[8];
 float  test_value2;
 offset = offsetof(calstruct, korr_A[3][0]);
 for (int i = 0; i < 8; i++) {
        myMem.get((LOCATION_CAL_LIPO + offset + sizeof(float)), mem_valuefloat);  
       test array[i] =  mem_valuefloat;  
 }
 test_value2 = korr_A[3][4];    -> corrpt data "-inf"
```

gigapod, 2025-07-28:
Hi @herbert131  -

We'll take a look at this. maybe this is triggering some issue with reading data spanning a page boundary.

Anything more to the above example? Or is it just write out the example struct with data, and read it back. 

Thanks
-Kirk

gigapod, 2025-07-29:
Hi @herbert131 -

I'm unable to reproduce the issue with a simple test sketch (I'll paste it below).  Note I'm. testing using our qwiic  EEPROM board (24XX512 IC). 

Do you have a simple example that replicates the issue? 

-Kirk

My test example (I'm testing using an RP2350 board):

```c++

#include <Wire.h>

#include "SparkFun_External_EEPROM.h" // Click here to get the library: http://librarymanager/All#SparkFun_External_EEPROM
ExternalEEPROM myMem;

// Test struct
typedef struct
{
    uint8_t version;     // 1 byte
    uint8_t reserved[3]; // 3 Byte  padding

    bool cal_A[4];      // 4 Byte
    bool cal_B[4];      // 4 Byte
    float korr_B[4][8]; // 128 Byte
    float korr_A[4][8]; // 128 Byte

    uint16_t reserved2; // 2 Byte
    uint16_t crc;       // 2 Byte
} calstruct;

calstruct srcCal = {.version = 1,
                    .reserved = {0, 0, 0},
                    .cal_A = {true, false, true, false},
                    .cal_B = {false, true, false, true},
                    .korr_B = {{0.0f}}, // Initialize to zero
                    .korr_A = {{0.0f}}, // Initialize to zero
                    .reserved2 = 11,
                    .crc = 22};

calstruct readCal = {.version = 0,
                     .reserved = {0, 0, 0},
                     .cal_A = {false, false, false, false},
                     .cal_B = {false, false, false, false},
                     .korr_B = {{0.0f}}, // Initialize to zero
                     .korr_A = {{0.0f}}, // Initialize to zero
                     .reserved2 = 0,
                     .crc = 0};

// How many records to write/read?
#define NUMBER_OF_RECORDS 300

bool testRecord(calstruct &record)
{
    bool status = true;
    for (int i = 0; i < 4; i++)
    {
        for (int j = 0; j < 8; j++)
        {
            if (record.korr_B[i][j] != srcCal.korr_B[i][j])
            {
                Serial.printf("\t Value Error. korr_b[%d][%d] = %f != %f\n\r", i, j, record.korr_B[i][j],
                              srcCal.korr_B[i][j]);

                status = false;
            }
            if (record.korr_A[i][j] != srcCal.korr_A[i][j])
            {
                Serial.printf("\t Value Error. korr_A[%d][%d] = %f != %f\n\r", i, j, record.korr_A[i][j],
                              srcCal.korr_A[i][j]);
                status = false;
            }
        }
    }
    // This function can be used to test if a record is valid
    // For now, we just check the version
    return status;
}

void setup()
{
    Serial.begin(115200);

    delay(2000);

    // delay(250); //Often needed for ESP based platforms
    Serial.println(F("-------------------------------------"));
    Serial.println(F("Qwiic EEPROM Structure Test example"));
    Serial.println(F("-------------------------------------"));

    Wire.begin();

    // Default to the Qwiic 24xx512 EEPROM: https://www.sparkfun.com/products/18355
    myMem.setMemoryType(512); // Valid types: 0, 1, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1025, 2048
    // The 512 is 2 by address, page size of 128 bytes
    // myMem.setPageSizeBytes(124);
    // myMem.setAddressBytes(2);

    if (myMem.begin() == false)
    {
        Serial.println(F("[ERROR] No memory detected. Freezing."));
        while (true)
            ;
    }
    Serial.println(F("[OK] Memory detected!"));

    Serial.println("");
    Serial.printf("Beginning EEPROM test... record size is %d bytes\n\r", sizeof(calstruct));
    Serial.println(F("Setting up source data arrays..."));

    for (int i = 0; i < 4; i++)
    {
        for (int j = 0; j < 8; j++)
        {
            srcCal.korr_B[i][j] = (float)(i + j + 1000.);
            srcCal.korr_A[i][j] = (float)(i - j - 1000.);
        }
    }

    Serial.println("");

    // Make sure we don't exceed the EEPROM size in our testing

    uint32_t totalSize = myMem.getMemorySizeBytes();
    Serial.printf("Total EEPROM size: %d bytes\n\r", totalSize);
    uint32_t totalRecords = totalSize / sizeof(calstruct);
    totalRecords = totalRecords > NUMBER_OF_RECORDS ? NUMBER_OF_RECORDS : totalRecords;
    Serial.printf("Total records available: %d\n\r", totalRecords);
    uint32_t modder = totalRecords / 100;

    // write out our source calibration data
    Serial.printf("Writing calibration data to EEPROM\t...");
    for (uint32_t i = 0; i < totalRecords; i++)
    {
        // Note - using crc as a record counter - one's based
        srcCal.crc = i + 1;                       // Set the record number in the crc field
        myMem.put(i * sizeof(calstruct), srcCal); // Write the struct to EEPROM at address
        if (i % modder == 0)
            Serial.print(".");
    }
    Serial.println("");
    Serial.println("");

    uint32_t success = 0;
    // now read it back
    Serial.printf("Reading calibration data from EEPROM\t...");
    for (uint32_t i = 0; i < totalRecords; i++)
    {
        myMem.get(i * sizeof(calstruct), readCal); // Read the struct from EEPROM at address
        if (readCal.crc != i + 1)
        {
            // Check if the record number matches
            Serial.printf("Bad ID - record: %d, expected: %d, got: %d. Read offset: %d!\n\r", i, i + 1, readCal.crc,
                          i * sizeof(calstruct));
        }
        else if (!testRecord(readCal))
            Serial.printf("Validation Error - record: %d!\n\r", i);
        else
            success++;
        if (i % modder == 0)
            Serial.print(".");
    }
    Serial.println("");
    Serial.println("");

    Serial.println(F("----------------------------------------------"));
    Serial.printf("Record read/write test of [%d] records complete.\n\r", totalRecords);

    Serial.printf("\n\rResults:  PASSED: %d, FAILED: %d\n\r", success, totalRecords - success);
    Serial.println(F("----------------------------------------------"));
}

void loop()
{
    delay(1000);
}

```

herbert131, 2025-08-14:
Thanks for your support, unfortunately it was my fault. A pointer was pointing out side its boundaries. It caused no issues so far but with the extension to more than 255 byte it caused unforseen behaviour

nseidle, 2025-08-14:
Thanks for letting us know! I'm going to close this issue. Please let us know if you see any other problems.

closed 2025-08-14 by nseidle